vikt0rm/dbus-shelly-1pm-pvinverter #1: Wrong values total production /consumption

frommelt5, 2022-04-18:
Hey, 
This piece if software is great but in my setup there is an issue with the total production and consumption values. Every morning at around 6 o'clock it is adding a huge amount of kWh, probably the value of the hole  last day.
The other values are always correct.
Setup: 2 Shelly 1pm on Hoymiles Inverters and Victron multiplus II (ESS Mode).
Any idea? What did I misconfigure? I think the values were correct before I started using this script 
[image]

vikt0rm, 2022-04-20:
Hi, 

yes i saw that issue as well. Need to take a closer look what exactly is going wrong there. I will fix that hopefully today. 
Thank you for the feedback!

BR,
Viktor

vikt0rm, 2022-04-21:
Issue is fixed with last [commit](https://github.com/vikt0rm/dbus-shelly-1pm-pvinverter/commit/39c38ce6392f7f2f40410d040254113930a17d03) on master.

frommelt5, 2022-04-23:
Hey,
It's working perfectly now. Thank you very much

vikt0rm, 2022-04-26:
Yes it worked some days without issues, so I thought its good to go. But yesterday I also had a use case where energy value again was 0 causing this issue (root cause). 

I try to invalidate the initial energy values and see if that helps.