vikt0rm/dbus-shelly-1pm-pvinverter #4: current.log is getting too big

e-i-k-e, 2022-08-11:
Hello,

I noticed that the "current.log" file has grown to over 300 Mb in the meantime. Is there a possibility to limit the log file or that it cleans itself?

[image]

iz3man, 2022-09-22:
This seems to be an issues with all those scripts based on the same work by @fabian-lauer ?!
I got the issue with th goe-charger script.

This happens whenever something goes wrong an floods the log.
It seems there is no log-rotate or some kind of logfile size check and auto dumping.

This is a serious issue, as, as soon as the file is >2Gb it can't be addressed by 32bit venusos anymore.

fabian-lauer, 2022-09-22:
Ok - got it …. So we need a proper logging framework.
Any recommendations for python?

iz3man, 2022-09-22:
Sorry. No idea. I just discovered the issue. :(

fabian-lauer, 2022-09-22:
Will try this one here after my vacation: https://www.blog.pythonlibrary.org/2014/02/11/python-how-to-create-rotating-logs/

currently no notebook in place 😎

iz3man, 2022-09-22:
Nice. Yeah logrotate - no idea why it seems not to be inplemented in venusOS?!
Enjoy your vacation - Viel Spass noch! :)

mascheihei, 2022-09-24:
I have rotatelog already integrated. Will make a fork and and this

iz3man, 2022-09-24:
Great! Will you post a link to the fork? Why not just update the original version? I think it's benefical (if not mandatory) for every user, as a not noticed error can render the whole partition unusable and unwriteable as it's 100% full.

mascheihei, 2022-09-24:
I have created a pull request for the original fork. viktOrm can merge the changes into his main. I saw your bugreport and changed it 3 days ago before installing on the Cerbo of my friend. 
However: https://github.com/mascheihei/dbus-shelly-1pm-pvinverter

e-i-k-e, 2022-12-22:
Is there a solution meanwhile?

My memory was full and VenusOS didn't run properly anymore because I had two log files with 2.2 and 3 GB each.

pau1phi11ips, 2022-12-22:
@e-i-k-e just delete the logs files via SSH

e-i-k-e, 2022-12-22:
@pau1phi11ips 

Logical. But this does not permanently solve the problem.

mascheihei, 2022-12-23:
@e-i-k-e and @pau1phi11ips 

I tried with a pull request but it seems vikt=rm is gone ... Therefore I removed my fork.

With these changes you will get rotating logfiles for 7 days. the older log-files will be deleted.

adjust  the following:

in line 5 after import logging add

from logging.handlers import TimedRotatingFileHandler

in main after line 194 insert log rotate handler and change logging.basic... as below.

log_rotate_handler = TimedRotatingFileHandler("%s/current.log" % (os.path.dirname(os.path.realpath(__file__))),
                                     when="d",
                                     interval=1,
                                     backupCount=7)
                                     
logging.basicConfig(format='%(asctime)s,%(msecs)d %(name)s %(levelname)s %(message)s',
        datefmt='%Y-%m-%d %H:%M:%S',
        level=logging.INFO,
        handlers=[
        logging.StreamHandler(),
        log_rotate_handler
    ])

mascheihei, 2022-12-23:
Ah not sure why: 
....   os.path.dirname(os.path.realpath(__file__) .....

above has changed

vikt0rm, 2023-07-05:
Sorry for being absent for so long time, now I started again maintaining the code. Since I finally got my Multiplus 2 5000, battery and final PV installation these days. 

Will add the rotating log to my implementation, that is very good idea! Thanks for the hint!

ramack, 2023-07-13:
Isn't there a somehow "standard" approach for logging (which would also end up in /var/log and probably using the strange timestamping you need tai64n for? - I didn't search but this driver is not the first one, so I guess the victron guys did their job here and you could "simply" reuse it.

mpvader, 2024-05-31:
Hello all, yes there is a standard approach, read here:

And this current.log file approach is causing our users quite some headaches, and support issues sent to our distributors, and so forth.

Please help sort this sooner rather than later - thanks. https://github.com/henne49/dbus-opendtu/issues/173