vikt0rm/dbus-shelly-1pm-pvinverter #5: Unstable with latest Venus OS v2.87 v2.90?

mawi42, 2022-09-16:
Hello,
I was using this nice feature for a while with Venus v2.84. However when I updated to v2.87 and v2.90 it observed that very often the shelly does not show the energy it is actually measures (if I access it directly I see the correct value). I am wondering if that is a general problem of latest firmwares of Venus OS or somehow my installation. Going back to v2.84 all is OK.

WHat kind of debug information you need to investigate into the problem?

ThomasLohner, 2023-03-09:
Hi @mawi42 For me this happened when the shelly was not available because of reboot after firmware update. Restarting the script helped. It seems to me that the loop gathering data gives up at some point. Do you see an error in the log file?

mawi42, 2023-03-09:
Hi Thomas,
to be honest I did not catch up on this topic anymore, I stick to the old firmware which runs fully fine for me. So I have no log available. If it is important for the development let me know than I will do some tests.

Bets regards

mawi