vikt0rm/dbus-shelly-1pm-pvinverter #6: how can this be adopted to shelly-3em?

iz3man, 2022-09-22:
My huawei 2000L inverter is a three phase device which is monitored my a Shelly 3EM.

How can this script be adopted to 3-phases. As it seems this is based on a 3-phase script anyway, so it should be doable?

Thanks

pau1phi11ips, 2022-09-22:
@iz3man see https://github.com/fabian-lauer/dbus-shelly-3em-smartmeter

iz3man, 2022-09-22:
I know that, and it's in use, but this is for a GRIDMETER?! I want it to show as SOLAR INVERTER. If I just start another instance it will show up as energy meter.

pau1phi11ips, 2022-09-22:
I think you need add `elf._dbusservice.add_path('/Role', 'pvinverter ')`

iz3man, 2022-09-22:
Thanks. That was one of the things. Needed to change device_id and some other things as well. But it was like 5 changes only, and it seems to work now :)

pau1phi11ips, 2022-09-23:
This can be closed I think @vikt0rm

closed 2022-09-23 by iz3man